DeMarcoLab/autolamella #23: Add expansion joints to protocol

bbarad, 2020-11-16:
I am currently experimenting with manually milling micro-expansion joints before running autolamella, and would be interested in including the feature in autolamella to accelerate the process - is this something that you would be interested in accepting as a pull request?

GenevieveBuckley, 2020-11-16:
Hi Ben! Yes, that is a useful feature I'd be happy to accept a pull request for.

I'd like to see:
1. The new feature is optional so users can choose whether or not they'd like to mill expansion joints, and
2. Test coverage for the new code

Please let me know if you have any questions I can help you with.

bbarad, 2020-11-16:
Awesome!

I think the best way to do this cleanly would be to work with the fiducials milling routine as a template, where it is set up w/ a user input `y/n` for each lamella during target selection. I am hoping that will also make it straightforward for me to figure out the testing based on the existing fiducials tests. If that approach sounds appropriate (as opposed to, for example, incorporating a "stage 0" of the milling protocols), I can get started on putting it together.

GenevieveBuckley, 2020-11-16:
Expansion joints are pretty quick to mill, I think that sounds like a good approach

GenevieveBuckley, 2020-12-20:
Hi Ben, just wanted to let you know that Alex & I have recently had a conversation with another group who is also interested in milling expansion joints. He's happy for me to put the two of you in touch to discuss further, so I've sent an email with those contact details.

bbarad, 2020-12-20:
Sounds great! Definitely happy to collaborate or assist w/ testing + integration if the other group is already further along than I am! The end of year deadlines have slowed me down but this is still something I'd like to enable (so far I have been doing a quick manual mill of joints during the target selection process, which is obviously non-ideal!).

GenevieveBuckley, 2020-12-20:
> Sounds great! Definitely happy to collaborate or assist w/ testing + integration if the other group is already further along than I am! The end of year deadlines have slowed me down but this is [...]

Excellent!

MoritzWM, 2022-03-21:
Hi!
Even though this thread is quite old: we implemented microexpansion joints (and some other small stuff) in [our fork](https://github.com/Chlanda-Lab/autolamella/blob/master/protocol_chlanda.yml). Would you still be interested? Has anyone else done work on this?

GenevieveBuckley, 2022-05-11:
Hi @MoritzWM! I don't have write access to this project anymore, since I'm no longer working for this lab group.  

I've just sent an email to the lab head to let him know about your comment. I'm not sure who gets notifications about this project now, or whether it's currently unmaintained.

patrickcleeve2, 2022-05-16:
Hi @MoritzWM , 

If you submit a pull request with your fork,  I will work on getting the changes merged. 

Thanks!

closed 2022-07-29 by patrickcleeve2